svartalf/rust-battery #6: Could not compile `battery`

Dentrax, 2019-02-25:
When i run `cargo build --release` I got this:

```
Compiling battery v0.6.0
   Compiling battery v0.6.0 (/home/dentrax/Projects/GitHub/rust-battery/battery)
   Compiling quote v0.6.11
   Compiling rand_chacha v0.1.1
   Compiling rand_pcg v0.1.2
   Compiling rand_os v0.1.2
   Compiling atty v0.2.11
   Compiling termion v1.5.1
error[E0658]: imports can only refer to extern crate names passed with `--extern` on stable channel (see issue #53130)
  --> /home/dentrax/.cargo/registry/src/github.com-1ecc6299db9ec823/battery-0.6.0/src/lib.rs:34:9
   |
31 | mod types;
   | ---------- not an extern crate passed with `--extern`
...
34 | pub use types::{Manager, Batteries, Battery, State, Technology};
   |         ^^^^^
   |
   = help: add #![feature(uniform_paths)] to the crate attributes to enable
note: this import refers to the module defined here
  --> /home/dentrax/.cargo/registry/src/github.com-1ecc6299db9ec823/battery-0.6.0/src/lib.rs:31:1
   |
31 | mod types;
   | ^^^^^^^^^^

error[E0658]: imports can only refer to extern crate names passed with `--extern` on stable channel (see issue #53130)
  --> battery/src/lib.rs:34:9
   |
31 | mod types;
   | ---------- not an extern crate passed with `--extern`
...
34 | pub use types::{Manager, Batteries, Battery, State, Technology};
   |         ^^^^^
   |
   = help: add #![feature(uniform_paths)] to the crate attributes to enable
note: this import refers to the module defined here
  --> battery/src/lib.rs:31:1
   |
31 | mod types;
   | ^^^^^^^^^^

   Compiling clap v2.32.0
error: aborting due to previous error

For more information about this error, try `rustc --explain E0658`.
error: Could not compile `battery`.
warning: build failed, waiting for other jobs to finish...
error: aborting due to previous error

For more information about this error, try `rustc --explain E0658`.
error: Could not compile `battery`.
warning: build failed, waiting for other jobs to finish...
error: build failed
```

**OS:** Arch Linux

svartalf, 2019-02-25:
Hey, @Dentrax! Thank you for report!

I have tested `battery` with Rust 1.32 and you're probably using Rust <= 1.31, where [uniform paths](https://github.com/rust-lang/rust/pull/56759) were not stabilized yet.

1e50430140fc3405c0dfe78329d0012d5e6d3663 should fix the problem, it is already in master branch.
Can you confirm that compilation is working now for you?

Dentrax, 2019-02-25:
@svartalf 

Thank you it is working fine now. :)

closed 2019-02-25 by Dentrax

svartalf, 2019-02-25:
I'm glad to hear that :)