svartalf/rust-battery #8: Mac OS version doesn't work correctly

non-descriptive, 2019-02-25:
https://monosnap.com/file/wSyg3psLsM7m7EsdWBLYm2LcttjbIf
compiled just fine but data a lot of error data.
```
System Software Overview:
 System Version:	macOS 10.14.3 (18D39a)
  Kernel Version:	Darwin 18.2.0
  Boot Volume:	Macintosh HD
  Boot Mode:	Normal

Hardware Overview:
  Model Name:	MacBook Pro
  Model Identifier:	MacBookPro13,2
  Processor Name:	Intel Core i5
  Processor Speed:	2,9 GHz
  Number of Processors:	1
  Total Number of Cores:	2
  L2 Cache (per Core):	256 KB
  L3 Cache:	4 MB
  Memory:	16 GB

Battery Information:

  Model Information:
  Serial Number:	******
  Manufacturer:	SMP
  Device Name:	******
  Pack Lot Code:	0
  PCB Lot Code:	0
  Firmware Version:	901
  Hardware Revision:	1
  Cell Revision:	3922
  Charge Information:
  Charge Remaining (mAh):	4123
  Fully Charged:	Yes
  Charging:	No
  Full Charge Capacity (mAh):	4203
  Health Information:
  Cycle Count:	50
  Condition:	Normal
  Battery Installed:	Yes
  Amperage (mA):	-1037
  Voltage (mV):	12916

System Power Settings:
  AC Power:
  System Sleep Timer (Minutes):	1
  Disk Sleep Timer (Minutes):	10
  Display Sleep Timer (Minutes):	10
  Wake on AC Change:	No
  Wake on Clamshell Open:	Yes
  Wake on LAN:	Yes
  AutoPowerOff Delay:	28800
  AutoPowerOff Enabled:	1
  DarkWakeBackgroundTasks:	1
  Display Sleep Uses Dim:	Yes
  GPUSwitch:	2
  Hibernate Mode:	3
  High Standby Delay:	86400
  PrioritizeNetworkReachabilityOverSleep:	0
  ProximityDarkWake:	1
  Standby Battery Threshold:	50
  Standby Delay:	10800
  Standby Enabled:	1
  TCPKeepAlivePref:	1
  Battery Power:
  System Sleep Timer (Minutes):	1
  Disk Sleep Timer (Minutes):	10
  Display Sleep Timer (Minutes):	2
  Wake on AC Change:	No
  Wake on Clamshell Open:	Yes
  AutoPowerOff Delay:	28800
  AutoPowerOff Enabled:	1
  Current Power Source:	Yes
  DarkWakeBackgroundTasks:	0
  Display Sleep Uses Dim:	Yes
  GPUSwitch:	2
  Hibernate Mode:	3
  High Standby Delay:	86400
  ProximityDarkWake:	0
  Reduce Brightness:	Yes
  Standby Battery Threshold:	50
  Standby Delay:	10800
  Standby Enabled:	1
  TCPKeepAlivePref:	1

Hardware Configuration:
  UPS Installed:	No

AC Charger Information:
  Connected:	No
  Charging:	No
```

svartalf, 2019-02-25:
Thank you for this bug-report, @non-descriptive! Could you also attach output from the `ioreg -n AppleSmartBattery -r -a` terminal command?

While information that you provided is very helpful, it does not show some keys used in energy calculation and it is making investigation a little bit harder.

non-descriptive, 2019-02-26:
Yeah, here you go
```<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<array>
	<dict>
		<key>AdapterInfo</key>
		<integer>0</integer>
		<key>Amperage</key>
		<integer>0</integer>
		<key>AppleRawCurrentCapacity</key>
		<integer>3938</integer>
		<key>AppleRawMaxCapacity</key>
		<integer>4119</integer>
		<key>AvgTimeToEmpty</key>
		<integer>65535</integer>
		<key>AvgTimeToFull</key>
		<integer>65535</integer>
		<key>BatteryData</key>
		<dict>
			<key>CycleCount</key>
			<integer>50</integer>
			<key>DesignCapacity</key>
			<integer>4315</integer>
			<key>QmaxCell0</key>
			<integer>22545</integer>
			<key>QmaxCell1</key>
			<integer>43281</integer>
			<key>QmaxCell2</key>
			<integer>32273</integer>
			<key>ResScale</key>
			<integer>32000</integer>
			<key>StateOfCharge</key>
			<integer>24576</integer>
			<key>Voltage</key>
			<integer>12818</integer>
		</dict>
		<key>BatteryFCCData</key>
		<dict>
			<key>DOD0</key>
			<integer>0</integer>
			<key>DOD1</key>
			<integer>0</integer>
			<key>DOD2</key>
			<integer>0</integer>
			<key>PassedCharge</key>
			<integer>0</integer>
			<key>ResScale</key>
			<integer>32000</integer>
		</dict>
		<key>BatteryInstalled</key>
		<true/>
		<key>BatteryInvalidWakeSeconds</key>
		<integer>30</integer>
		<key>BatterySerialNumber</key>
		<string>D867097H21XHDTDA1</string>
		<key>BootPathUpdated</key>
		<integer>1551086507</integer>
		<key>CellVoltage</key>
		<array>
			<integer>4273</integer>
			<integer>4273</integer>
			<integer>4273</integer>
			<integer>0</integer>
		</array>
		<key>ChargerData</key>
		<dict>
			<key>ChargingCurrent</key>
			<integer>0</integer>
			<key>ChargingVoltage</key>
			<integer>64050</integer>
			<key>NotChargingReason</key>
			<integer>4</integer>
		</dict>
		<key>CurrentCapacity</key>
		<integer>3938</integer>
		<key>CycleCount</key>
		<integer>50</integer>
		<key>DesignCapacity</key>
		<integer>4315</integer>
		<key>DesignCycleCount70</key>
		<integer>0</integer>
		<key>DesignCycleCount9C</key>
		<integer>1000</integer>
		<key>DeviceName</key>
		<string>bq20z451</string>
		<key>ExternalChargeCapable</key>
		<true/>
		<key>ExternalConnected</key>
		<true/>
		<key>FirmwareSerialNumber</key>
		<integer>1</integer>
		<key>FullPathUpdated</key>
		<integer>1551172742</integer>
		<key>FullyCharged</key>
		<true/>
		<key>IOGeneralInterest</key>
		<string>IOCommand is not serializable</string>
		<key>IOObjectClass</key>
		<string>AppleSmartBattery</string>
		<key>IOObjectRetainCount</key>
		<integer>6</integer>
		<key>IORegistryEntryID</key>
		<integer>4294967994</integer>
		<key>IORegistryEntryName</key>
		<string>AppleSmartBattery</string>
		<key>IOReportLegend</key>
		<array>
			<dict>
				<key>IOReportChannelInfo</key>
				<dict>
					<key>IOReportChannelUnit</key>
					<integer>0</integer>
				</dict>
				<key>IOReportChannels</key>
				<array>
					<array>
						<integer>7167869599145487988</integer>
						<integer>6460407809</integer>
						<string>BatteryCycleCount</string>
					</array>
				</array>
				<key>IOReportGroupName</key>
				<string>Battery</string>
			</dict>
		</array>
		<key>IOReportLegendPublic</key>
		<true/>
		<key>IOServiceBusyState</key>
		<integer>0</integer>
		<key>IOServiceBusyTime</key>
		<integer>42558</integer>
		<key>IOServiceState</key>
		<integer>30</integer>
		<key>InstantAmperage</key>
		<integer>0</integer>
		<key>InstantTimeToEmpty</key>
		<integer>65535</integer>
		<key>IsCharging</key>
		<false/>
		<key>LegacyBatteryInfo</key>
		<dict>
			<key>Amperage</key>
			<integer>0</integer>
			<key>Capacity</key>
			<integer>4119</integer>
			<key>Current</key>
			<integer>3938</integer>
			<key>Cycle Count</key>
			<integer>50</integer>
			<key>Flags</key>
			<integer>5</integer>
			<key>Voltage</key>
			<integer>12818</integer>
		</dict>
		<key>Location</key>
		<integer>0</integer>
		<key>ManufactureDate</key>
		<integer>19044</integer>
		<key>Manufacturer</key>
		<string>SMP</string>
		<key>ManufacturerData</key>
		<data>
		AAAAAAkBAAE5IgAAAzMxMQMwMDMDQVRMAUwA
		</data>
		<key>MaxCapacity</key>
		<integer>4119</integer>
		<key>MaxErr</key>
		<integer>1</integer>
		<key>OperationStatus</key>
		<integer>58435</integer>
		<key>PackReserve</key>
		<integer>125</integer>
		<key>PermanentFailureStatus</key>
		<integer>0</integer>
		<key>PostChargeWaitSeconds</key>
		<integer>120</integer>
		<key>PostDischargeWaitSeconds</key>
		<integer>120</integer>
		<key>Temperature</key>
		<integer>3102</integer>
		<key>TimeRemaining</key>
		<integer>0</integer>
		<key>UserVisiblePathUpdated</key>
		<integer>1551173222</integer>
		<key>Voltage</key>
		<integer>12818</integer>
	</dict>
</array>
</plist>
```

svartalf, 2019-02-26:
Great, thank you!

This is probably happening because of the `amperage` value overflow, same as in #11. I'll investigate it further

svartalf, 2019-02-27:
Okay, I have managed to reproduce and fix that bug, `0.6.1` version is [published](https://crates.io/crates/battery) already.

closed 2019-02-27 by svartalf

non-descriptive, 2019-02-28:
I think you should also update README since Mac OS confirmed working

svartalf, 2019-02-28:
I'm rewriting Mac OS implementation right now within #2, after that I'll probably know which Mac OS version should be considered as minimal.
Thanks for a reminder :)

inferiorhumanorgans, 2022-04-01:
FWIW I'm seeing something similar.

```
$ uname -v
Darwin Kernel Version 18.7.0: Tue Jun 22 19:37:08 PDT 2021; root:xnu-4903.278.70~1/RELEASE_X86_64
$ cargo run --example simple
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `target/debug/examples/simple`
Battery { impl: MacOSDevice { source: PowerSource { io_object: IoObject(2563) } }, vendor: Some("DP"), model: Some("bq20z451"), serial_number: Some("C018274DT7WFWLMFS"), technology: Unknown, state: Charging, capacity: 0.59497434, temperature: Some(303.78998 K^1), percentage: 0.9808025, cycle_count: Some(1213), energy: 231229.27 m^2 kg^1 s^-2, energy_full: 235755.19 m^2 kg^1 s^-2, energy_full_design: 396244.3 m^2 kg^1 s^-2, energy_rate: 7.807213 m^2 kg^1 s^-3, voltage: 12.5720005 m^2 kg^1 s^-3 A^-1, time_to_full: Some(720.0 s^1), time_to_empty: None }
```

Note that the menu bar shows 100% and charged and the LED on the charger itself is green and the connector is cool to the touch (suggesting it's not charging).  The percent of charge reported by the crate is slowly approaching 100%.  While it's nice to have access to (potentially) more accurate information it would also be nice to mimic what the OS and charger are reporting.