svartalf/rust-battery #12: Debian packaging for Pop!_OS

mmstick, 2019-02-26:
Whenever you're ready, I'll package and add this to our Pop repository. This seems like a very useful tool for us.

svartalf, 2019-02-26:
You mean CLI app, right?

Do you have any integration process documentation or anything similar worth to read out of curiosity?

mmstick, 2019-02-26:
Yes, the CLI application, or any available user-facing binary project. A GTK application would be nice to have, for example.

There's no documentation for the packaging process. We set up cargo packages to create a source tarball that contains an inner vendor tarball, so that the build system can build the Cargo project without Internet access in Launchpad. No work has to be done on your part.

svartalf, 2019-02-26:
@mmstick, okay, got it, thank you.

Yet as I was said a couple times before on reddit, CLI application was basically a proof of concept and from my point of view it is not ready to be used everywhere.
First of all, it would be nice to get a community feedback (as in #13, for example), stabilize it a little and move on then.

Any way, this idea sounds amazing :)