svartalf/rust-battery #14: Overflow panic when pretty printing Battery struct.

bddap, 2019-02-26:
Setup:

```rust
fn main() {
    let batteries = battery::Manager::new().iter().collect::<Vec<_>>();
    println!("{:#?}", batteries);
}
```

Error 😨:

```
[
    Battery {
        vendor: Some(
            "SMP"
        ),
        model: Some(
            "bq20z451"
        ),
        serial_number: Some(
            "D866466Q2BPFWLKFQ"
        ),
        technology: Unknown,
        state: Full,
        capacity: 0.0,
        temperature: Some(
            30.11
thread 'main' panicked at 'attempt to multiply with overflow', /Users/a/.cargo/registry/src/github.com-1ecc6299db9ec823/battery-0.6.0/src/platform/macos/device.rs:128:10
note: Run with `RUST_BACKTRACE=1` for a backtrace.
        )
```

OS: macOS High Sierra 10.13.6 17G5019 x86_64 
Host: MacBookPro11,4

bddap, 2019-02-26:
Nice crate btw.

svartalf, 2019-02-26:
Thanks :)

Are you using version from the crates.io or from the github? I think this was fixed in #10, but was not published yet.

bddap, 2019-02-26:
I was using the version on crates.io.

svartalf, 2019-02-26:
Any chance you can test latest master?

bddap, 2019-02-26:
Master works fine:

```
[
    Battery {
        vendor: Some(
            "SMP"
        ),
        model: Some(
            "bq20z451"
        ),
        serial_number: Some(
            "D866466Q2BPFWLKFQ"
        ),
        technology: Unknown,
        state: Full,
        capacity: 0.0,
        temperature: Some(
            30.17
        ),
        percentage: 99.36493,
        cycle_count: Some(
            112
        ),
        energy: 105011352,
        energy_full: 105682516,
        energy_full_design: 113000785,
        energy_rate: 0,
        voltage: 12907,
        time_to_full: None,
        time_to_empty: None
    }
]
```

svartalf, 2019-02-26:
Yay! Unfortunately, I see that you are also affected by #8 bug, so I'll leave this issue opened for a while.

Going to try to fix it tomorrow and post a comment here.

svartalf, 2019-02-27:
Since #8 was closed, I'm closing this issue too.

Fixed versions were published to crates.io

closed 2019-02-27 by svartalf