svartalf/rust-battery #24: Support UPSs handled by APC

boomshroom, 2019-03-16:
As someone whose computer is connected via USB to an external power supply, I would love to be able to monitor its status from within battop. However it doesn't appear in the sysfs and as such is invisible to this library. Being able to support more types of batteries and power supplies should be a worthy goal for this type of library, but I don't know how much effort it would take to implement.

svartalf, 2019-03-31:
Hey, @boomshroom!

I seriously doubt that this is the good idea, sorry. `battery` is nothing more than an abstraction over OS interfaces, and if target platform does not have the driver for your power supply, this crate is useless in that case.

Also, I do not want to commit myself to support all these various and rare devices for all supported platforms; that would be a hell of a burden.

boomshroom, 2019-03-31:
I do have a driver in the form of [apcupsd](https://wiki.archlinux.org/index.php/APC_UPS). I can access the statistics through `apcaccess status`. Unfortunately it doesn't seem to come with a library that can be linked to and needs the output to be parsed in order to be used, nor does it populate `/sys/class/power_supply`.

svartalf, 2019-03-31:
Oh, okay, I've probably misunderstood you.

As far as I see from manual, the only options to get data from the `apcupsd` without re-implementing it fully, is to run shell command `apcaccess status` or try to make the HTTP request to a daemon CGI interface (if it is even enabled).

I'm going to re-open that issue, but with a low priority, because ADP information is not supported at all right now and it should be done first.