svartalf/rust-battery #25: About how to build and battery info scraping

Dentrax, 2019-03-31:
Hey,

I want to use this awesome project in [polybar](https://github.com/jaagr/polybar) via shell script. But i can't figure it out how to compile **cli** or **ffi**. I installed **battop** via `yay battop`. It works perfectly.

Actually, i couldn't see a installation guide or it could be a not-well documented. Simply, 
1. When i run this: `cargo install battery-ffi`

Returns:

```
Updating crates.io index
  Installing battery-ffi v0.7.0
error: specified package has no binaries
```

2. `battop` is not supporting output args like this:

`battop --time-to-empty`

It may be really really life saver.

3. `cargo build --release`

Returns:

```
error: failed to resolve patches for `https://github.com/rust-lang/crates.io-index`

Caused by:
  patch for `battery` in `https://github.com/rust-lang/crates.io-index` did not resolve to any crates. If this is unexpected, you may wish to consult: https://github.com/rust-lang/cargo/issues/4678
```

The bare-minimum thing I want is looks like:

`$ battery-cli --time-to-empty`

Consequently, I want to write the battery's `battery_get_time_to_empty` output to the polybar. I am trying to figure it out how to write a script that scrapes about battery informations.

svartalf, 2019-03-31:
Hey, @Dentrax!

Here is a deal:

`battery-ffi` is a thin interface for communication with programs written in other languages; for example, you can write a Python script, which will load the compiled `battery_ffi.so` library, call the `battery_get_time_to_empty` function and get back the Python `int` type.
There are examples for C and Python exists, you may want to check them out: https://github.com/svartalf/rust-battery/tree/master/battery-ffi/examples

Also, please note that `battery-ffi` is a library and not a binary application, so you can't `cargo install` it.

`battop` is the TUI interface, like [vim](https://www.vim.org) or [midnight commander](https://midnight-commander.org). It sounds like a good idea to implement one-time output into JSON or plain text, but at the moment it just does not have that ¯\_(ツ)_/¯

What I can suggest right now: take the [example](https://github.com/svartalf/rust-battery/blob/master/battery/examples/simple.rs) and write your own application which will output data in the required format.
Since it would be the shell script, you just need to create the `Manager` struct, iterate over available batteries and write the data into stdout.
You can also find multiple example here [at the docs.rs](https://docs.rs/battery/*/battery/units/index.html)

P.S. I'm also glad to hear that AUR package is working correctly :)

Dentrax, 2019-03-31:
@svartalf Thank you so much! So in this case, i should write my own wrapper using the `battery-ffi` lib. Previously repo had the `battery-cli` but i guess it no longer exists.

My second question is: can i use compiled `battery_ffi.so` in my `.sh` scripts?

svartalf, 2019-03-31:
@Dentrax, there are two options here: either you can write wrapper in Rust or in any other language.

In the former case you will need the `battery` crate only: add it into `Cargo.toml` dependencies block and call as any other crate library.

In the latter case you will need to:
 1. Compile `battery-ffi` crate
 2. Get the compiled `.so` file (and `.h` file if you want to)
 3. Link or load them into your wrapper
 4. Call any function [available](https://docs.rs/battery-ffi/0.7.0/battery_ffi/).

I doubt that there is any good way exists to call FFI functions directly from the shell scripts, so you'll probably need to call your wrapper binary instead.

Dentrax, 2019-03-31:
Thank you i will try that. And finally, how can i compile the `battery-ffi` crate? I am asking this simply question because `cargo build --release` not works. When i run `rustc build.rs`, it creates `build` file. 

`build: ELF 64-bit LSB pie executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 3.2.0, BuildID[sha1]=522085c26c0bbc8efc450c3c2e2a9fa3f83b645b, not stripped`

Is that correct? Am i missing something? (Yet another newbie question :)) Ty.

svartalf, 2019-03-31:
@Dentrax what is an error for `cargo build --release`?
Any chance it fails with the following message?

```
error: failed to resolve patches for `https://github.com/rust-lang/crates.io-index`

Caused by:
  patch for `battery` in `https://github.com/rust-lang/crates.io-index` did not resolve to any crates. If this is unexpected, you may wish to consult: https://github.com/rust-lang/cargo/issues/4678
```

Dentrax, 2019-04-01:
@svartalf Exactly same error with mine. How can i build that with correct way?

svartalf, 2019-04-01:
@Dentrax there is a "[0.7.1](https://github.com/svartalf/rust-battery/tree/0.7.1)" branch with a fix for this bug.

I'm planning to merge it into the `master` tomorrow and publish properly. Before that, you can use either this git branch or the pre-compiled `.so` file from the [previous release](https://github.com/svartalf/rust-battery/releases/tag/0.7.0).

svartalf, 2019-04-06:
Hey, @Dentrax, fyi, I've published `0.7.1` version to the [crates.io](https://crates.io/crates/battery).
Sorry it took so long.

Do you still need this issue to be opened?

Dentrax, 2019-04-08:
@svartalf 

Thank you, now it's compiling without any error. :)

closed 2019-04-08 by Dentrax