svartalf/rust-battery #34: Watch for updates

Gerschtli, 2019-09-14:
Hello everyone, nice project you created here!

Is it planned to provide a watcher-like functionality? For my package [dwm-status](https://github.com/Gerschtli/dwm-status) I build a watcher based on dbus events, maybe you are interested in adding a similar API method.

svartalf, 2019-09-14:
Hey, @Gerschtli!

There are no plans to make anything like this right now, as there is no such a notification mechanism for all platforms currently supported (note that Linux implementation is using `sysfs` instead of `uevent`+`dbus`).

It is possible to make this "watcher" in a polling manner and just re-read all data each `n` time, but it looks like a easy task to implement by yourself ([`Manager::refresh`](https://docs.rs/battery/0.7.4/battery/struct.Manager.html#method.refresh) could help with that).

Gerschtli, 2019-09-15:
Ah okay, the polling way is not an acceptable way for my project because of its ineffeciency and inaccuracy. But thank you for your quick response!

closed 2019-09-15 by Gerschtli