svartalf/rust-battery #40: Handle missing energy_full_design value for Linux

svartalf, 2019-11-01:
According to https://github.com/starship/starship/issues/613#issuecomment-548873632, there exist a possibility when both `energy_full_design` and `charge_full_design` files are missing for the battery device, so fetching fails with an error here: https://github.com/svartalf/rust-battery/blob/08b2ad579d135dbe48d52ff31a35df307c173eb6/battery/src/platform/linux/sysfs/source.rs#L183

`upower` just falls back to `0.0` value, but that might affect `state_of_health` 

https://github.com/svartalf/rust-battery/blob/08b2ad579d135dbe48d52ff31a35df307c173eb6/battery/src/platform/linux/sysfs/source.rs#L137

and `energy` parameters:

https://github.com/svartalf/rust-battery/blob/08b2ad579d135dbe48d52ff31a35df307c173eb6/battery/src/platform/linux/sysfs/source.rs#L168

svartalf, 2019-11-02:
@pizzafox, I dropped a quick fix, would you be able to test it with the `starship` of yours?

You can do that by adding the following lines at the end of the `Cargo.toml` file and rebuilding the `starship` project:

```toml
[patch.crates-io]
battery = { git = "https://github.com/svartalf/rust-battery.git", branch = "issue-40" }
```

Or with the patch:

```patch
diff --git a/Cargo.toml b/Cargo.toml
index 2b5ee64..60c16ca 100644
--- a/Cargo.toml
+++ b/Cargo.toml
@@ -57,3 +57,6 @@ lto = true
 [[bin]]
 name = "starship"
 path = "src/main.rs"
+
+[patch.crates-io]
+battery = { git = "https://github.com/svartalf/rust-battery.git", branch = "issue-40" }
```

pizzafox, 2019-11-03:
This fixes the issue! The battery is displaying properly in [Starship](https://starship.rs).

closed 2019-11-26 by svartalf

pizzafox, 2020-04-24:
I just upgraded to WSL 2 which has [a much better integration to the system APIs](https://docs.microsoft.com/en-us/windows/wsl/wsl2-about#full-system-call-compatibility)... Including battery!

This bug is no longer occurring on for me despite not upgrading rust-battery to a patched version.

svartalf, 2020-04-24:
@pizzafox thanks for an update, good to know!

commit 9c6e1bb2 (2019-11-26): Falling back to zero value if not able to read energy_full_design value for Linux (closes #40)