svartalf/rust-battery #44: No license files on crates.io

tim77, 2019-11-25:
There LICENSE-APACHE and LICENSE-MIT files on github, but no license files on crates.io. Could you add them? This is mandatory for some Linux distros if we want to package rust-battery for official repos.

svartalf, 2019-11-26:
Yeah, sure thing, I was planning to do a release soon and can add them too.

I wonder if there maybe some checklist on that kind of stuff? Debian [packaging policy](https://wiki.debian.org/Teams/RustPackaging/Policy) does not mention that, for example

tim77, 2019-11-26:
Trying to package and push `rust-battery` in official [Fedora](https://docs.fedoraproject.org/en-US/packaging-guidelines/LicensingGuidelines/#_license_text) repos. :) Thank you!

svartalf, 2019-11-26:
Alright, I added the license file in 343b8693dc27cc68e0daa7808b40aded36c77757, they are properly added now (checked with the `cargo package` command):

```
~/projects/rust-battery/target/package/battery-0.7.4$ ls -1
build.rs
Cargo.lock
Cargo.toml
Cargo.toml.orig
examples
LICENSE-APACHE
LICENSE-MIT
README.md
src
target
```

Is there anything else we can do for packaging it into the Fedora repos? If everything looks good, I can publish a new version

tim77, 2019-11-26:
Thanks a lot. Because there is a lot different MIT licenses license file is needed. But not no need to rush and publish when you have time for this. Thanks again. Otherwise LGTM. Just some Chore like:
```diff
 [dependencies.uom]
-version = "^0.23"
+version = "^0.26"
```

svartalf, 2019-11-26:
`uom` dependency was updated in #41 already, it should be good:

https://github.com/svartalf/rust-battery/blob/452897adaaa23eea34585bfbd229d3a501ea50d3/battery/Cargo.toml#L23

And I also need to publish fix for #40, as it affects downstream packages, so I'll publish new version in a few hours.

svartalf, 2019-11-26:
Published as a [0.7.5](https://github.com/svartalf/rust-battery/releases/tag/0.7.5) version

closed 2019-11-26 by svartalf