svartalf/rust-battery #57: WSL support.

leglesslamb, 2020-03-23:
`rust-battery` doesn't yet work in WSL. It isn't detecting any batteries.

svartalf, 2020-03-23:
Oh, that's bad, thanks for a bug report!
I'll try to check that tomorrow. In a meanwhile, can you provide an output of the following command, executed in the WSL env?

```
 find /sys/class/power_supply/battery/ -name '*' -print -exec cat {} \;
```

leglesslamb, 2020-03-23:
Sure thing!

```sh
>find /sys/class/power_supply/battery/ -name '*' -print -exec cat {} \;
/sys/class/power_supply/battery/
cat: /sys/class/power_supply/battery/: Operation not permitted
/sys/class/power_supply/battery/capacity
98
/sys/class/power_supply/battery/charge_counter
3001
/sys/class/power_supply/battery/current_now
0
/sys/class/power_supply/battery/health
Good
/sys/class/power_supply/battery/present
1
/sys/class/power_supply/battery/status
Not charging
/sys/class/power_supply/battery/technology
Li-ion
/sys/class/power_supply/battery/temp
258
/sys/class/power_supply/battery/type
Battery
/sys/class/power_supply/battery/voltage_now
12659000
```

I can help with this issue, if you'd like.

svartalf, 2020-03-23:
Based on the files available it should work as expected, as it has the same files as in #40 case: https://github.com/svartalf/rust-battery/blob/eb7727bdf562382e074ea099c3dfcdfb8b1e71d3/battery/src/platform/linux/tests/issue_40.rs#L17-L28

@leglesslamb if you want to hack on this issue, feel free to do that!
I would start with a test case, similar to that one from above and based on your data.

leglesslamb, 2020-03-23:
I see. I was just getting `None` for many of the values when I used the example from your README.md.

```sh
Battery #0:
Vendor: None
Model: None
State: Unknown
Time to full charge: None
```
 
But actually using the crate in code, it seems to work. Thanks!

closed 2020-03-23 by leglesslamb