svartalf/rust-battery #62: failed to download `uom v0.26.0`

mkoakintomiwa, 2020-06-20:
Operating system: Windows.
After adding 

[dependencies]
battery = "0.7.5"

to my Cargo.toml and running 'cargo build', I get the below error:

error: failed to download `uom v0.26.0`

Caused by:
  unable to get packages from source

Caused by:
  failed to parse manifest at `C:\Users\MIS\.cargo\registry\src\github.com-1ecc6299db9ec823\uom-0.26.0\Cargo.toml`

Caused by:
  could not parse input as TOML

Caused by:
  unexpected character found: `\u{0}` at line 1 column 1

svartalf, 2020-06-20:
Hi, @mkoakintomiwa!

I don't think this is related to `battery` or `uom` at all; seems like your `cargo` failed to download `uom` crate properly.
Ensure that you are using modern enough Rust and cargo (`battery` requires at least Rust 1.36) and if that `Cargo.toml` manifest from your error message looks malformed, I suggest to remove all subfolder from the `C:\Users\MIS\.cargo\registry\src\` path.

mkoakintomiwa, 2020-06-20:
The build is now successful after removing all subfolders from the `C:\Users\MIS\.cargo\registry\src\ path`.

closed 2020-06-20 by mkoakintomiwa