svartalf/rust-battery #63: Unable to access battery information

mkoakintomiwa, 2020-06-20:
`Operating system: Windows`

I got `Unable to access battery information
Error: Error { source: Kind(InvalidData), description: Some("Device rate value is unknown") }
error: process didn't exit successfully: `target\debug\battery.exe` (exit code: 1)` When i ran this: 
```
extern crate battery;

use std::io;
use std::thread;
use std::time::Duration;

fn main() -> battery::Result<()> {
    let manager = battery::Manager::new()?;
    let mut battery = match manager.batteries()?.next() {
        Some(Ok(battery)) => battery,
        Some(Err(e)) => {
            eprintln!("Unable to access battery information");
            return Err(e);
        }
        None => {
            eprintln!("Unable to find any batteries");
            return Err(io::Error::from(io::ErrorKind::NotFound).into());
        }
    };

    loop {
        println!("{:?}", battery);
        thread::sleep(Duration::from_secs(1));
        manager.refresh(&mut battery)?;
    }
}
```

svartalf, 2020-06-27:
Hi, @mkoakintomiwa! Thank you for the bug report!

I managed to write a fix for it, can you test it? It's in the git branch right now, so you need to  **temporary** replace your `battery` dependency in `Cargo.toml` with the following dependency:

```toml
[dependencies]
battery = { git = "https://github.com/svartalf/rust-battery.git", branch = "issue-63" }
```

From what I understand, it can be reproduced when your laptop battery is fully charged and AC is plugged on, so your battery neither charged nor discharged.

svartalf, 2020-07-01:
I'm going to assume that 0d7b9e4 fixes this bug, so this issue is closed now. Thank you once again for the bug report!

commit 0d7b9e42 (2020-07-01): Zero energy rate is not considered as an error for Windows (closes #63)
closed 2020-07-01 by svartalf