svartalf/rust-battery #66: Energy rate miscalculation

mkoakintomiwa, 2020-07-01:
Operating system: `windows`
First of all, Thanks for always being there svartalf. Please help me with this one too.
```
extern crate uom;
extern crate math;

use math::round::half_to_even;

fn main() -> Result<(), battery::Error> {
    let manager = battery::Manager::new()?;

    for (_idx, maybe_battery) in manager.batteries()?.enumerate() {
        let battery = maybe_battery?;
        println!("Percentage charge: {:?}%",(half_to_even(f64::from(f32::from(battery.state_of_charge()))*100.0,0) as i32));
        println!("Time to full: {:?}", battery.time_to_full());
        println!("Time to empty: {:?}", battery.time_to_empty());
        println!("Energy rate: {:?}", battery.energy_rate());
        println!("State: {:?}", battery.state());
        println!("");
    }

    Ok(())
}
```

I got 

```
Percentage charge: 88%
Time to full: None
Time to empty: None
Energy rate: 0.0 m^2 kg^1 s^-3
State: Discharging
```

svartalf, 2020-07-02:
Hi, @mkoakintomiwa!

Without knowing current energy rate it is impossible to calculate `time_to_full` and `time_to_empty` values (and it goes to zero because of #63 fix).

It seems that you are hitting the following case from the [`BATTERY_STATUS` structure documentation at MSDN](https://docs.microsoft.com/en-us/windows/win32/power/battery-status-str#members) and your battery driver can't report it:

> **Rate**
>
>  If the rate is unavailable, this member is `BATTERY_UNKNOWN_RATE`. If the state of the battery or power source changes, the rate may become available.

I'm not actually sure if that's a case, because I can't reproduce your problem so far, so it is hard to say what the exact issue.
Yet, it seems that another option to fetch this information is to [query](https://docs.microsoft.com/en-us/windows/win32/power/ioctl-battery-query-information) the `BatteryEstimatedTime` value and use it instead.

Let me try to write a fix and maybe we will try to test it later then?

mkoakintomiwa, 2020-07-02:
Okay @svartalf , that is fine by me. I will wait for  the fix. Thanks alot.