svartalf/rust-battery #91: Could we get a release with a bumped version of nix?

ClementTsang, 2021-10-11:
One of the dependencies, nix 0.19, has [a security vulnerability](https://rustsec.org/advisories/RUSTSEC-2021-0119.html) and 0.19 does not have a patch with the fix, with only versions 0.20 and greater having patches to fix the vuln.

Would be great if we could get a dependency version bump + release to address this. I [tried updating nix to 0.22](https://github.com/ClementTsang/rust-battery) and it seems to work fine (0.23+ uses a newer version of bitflags which might problems for others).

rtzoeller, 2022-04-24:
Note that nix 0.24.1 is now out. The 0.24 release splits nix into features, so I'd recommend updating to that and enabling just the features you need to improve compile times (setting `default-features = false` along the way).

ClementTsang, 2022-05-06:
Anyone coming across this issue may want to consider migrating to https://crates.io/crates/starship-battery instead.