svartalf/rust-battery #94: Termux(Android) Support

bevsxyz, 2022-05-12:
Would it be feasible to have support for Android through [termux](https://termux.org/)?

bevsxyz, 2022-05-12:
```bash
error: Support for this target OS is not implemented yet!
        You may want to create an issue: https://github.com/svartalf/rust-battery/issues/new
  --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/battery-0.7.8/src/platform/mod.rs:27:9
   |
27 | /         compile_error!("Support for this target OS is not implemented yet!\n \
28 | |             You may want to create an issue: https://github.com/svartalf/rust-battery/issues/new");
   | |__________________________________________________________________________________________________^

error[E0432]: unresolved import `crate::platform::Device`
 --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/battery-0.7.8/src/types/battery.rs:5:5
  |
5 | use crate::platform::Device;
  |     ^^^^^^^^^^^^^^^^^^^^^^^ no `Device` in `platform`

error[E0432]: unresolved import `crate::platform::Iterator`
 --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/battery-0.7.8/src/types/iterator.rs:3:5
  |
3 | use crate::platform::Iterator as PlatformIterator;
  |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `Iterator` in `platform`

error[E0432]: unresolved import `crate::platform::Iterator`
 --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/battery-0.7.8/src/types/manager.rs:5:5
  |
5 | use crate::platform::Iterator as PlatformIterator;
  |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `Iterator` in `platform`

error[E0432]: unresolved import `crate::platform::Manager`
 --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/battery-0.7.8/src/types/manager.rs:6:5
  |
6 | use crate::platform::Manager as PlatformManager;
  |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `Manager` in `platform`

For more information about this error, try `rustc --explain E0432`.
error: could not compile `battery` due to 5 previous errors
```

svartalf, 2022-05-30:
Not in the roadmap, see https://github.com/svartalf/rust-battery/issues/33

closed 2022-05-30 by svartalf